looxonline/Marlin #1: [BUG] Wrong nozzle position when running Z Offset

Ejay-3D, 2020-11-14:
### Bug Description

When Running Z Offset The nozzle doesn't place itself at center of the bed ( Where the probe made its probe. 
Instead it goes more back left . 

### My Configurations

B1 ABL / B1 ABL HIGH RES Branch . 

### Steps to Reproduce

1. Go To ABL 
2. Click on Z Offset to turn it on . 

**Expected behavior:** 

Nozzles should be positioned at center of the bed when adjusting Z Offset 

**Actual behavior:** [What actually happens]
Nozzles is to on the back and on the left of the bed ( Not in the center  ) 

#### Additional Information

[video-1603372394.mp4.zip](https://github.com/looxonline/Marlin/files/5541454/video-1603372394.mp4.zip)

* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

looxonline, 2020-11-17:
According to the source code, your expected behavior is not how the standard BTT firmware even operates. Please install the standard BTT firmware and report back on how it responds when using this function.

Ejay-3D, 2020-11-17:
I use to have Marlin 2.0.6 installed on the machine and the nozzle position was at center when running Z offset

looxonline, 2020-11-17:
The nozzle position for offset is set by this code on the TFT: 
`/* Enable probe offset */
void probeOffsetEnable(void)
{
  probe_offset_enabled = true;

  probeHeightEnable();                                     // temporary disable software endstops

  // Z offset gcode sequence start
  mustStoreCmd("G28\n");                                   // home printer

  probeHeightStop();                                       // raise nozzle

  mustStoreCmd("G91\n");                                   // set relative position mode
  mustStoreCmd("G1 X%.2f Y%.2f F%d\n",
    getParameter(P_PROBE_OFFSET, X_STEPPER),
    getParameter(P_PROBE_OFFSET, Y_STEPPER),
    infoSettings.axis_speed[infoSettings.move_speed]);     // move nozzle to X and Y probing coordinates and set feedrate

  probeHeightStart();                                      // lower nozzle to Z0 point
}`

This means that the printer will home and then move the nozzle offset distance from the home point in order to set the z offset. This is a function of the TFT and not the marlin build. Please reinstall the 2.0.6 build and film the response.

Ejay-3D, 2020-11-17:
It look like the value are simply inversed , using the same value but reversed ( negative to positive , or positive to negative ) that would place the nozzle at the right position .

looxonline, 2020-11-17:
Yes but this is a function of the TFT firmware and not Marlin. Once again, the TFT firmware shows that the nozzle will never be in the center of the bed when using the function so I am very interested if your Marlin version made a difference because then it means that I may have misunderstood the TFT firmware. Please therefore take a video of the version of firmware where you mentioned it went to the center of the bed.

looxonline, 2020-11-20:
I have identified that there is a bug in either the TFT or Marlin that is causing inverted axes. If you send the commands manually it works: G28 G91 G1 X24 Y-46 F3500. These are the exact commands that the TFT should be sending to Marlin yet something is happening where either the TFT is sending the wrong G1 or Marlin is inverting it. I am going to logic analyze the UART line this weekend to investigate.

looxonline, 2020-11-26:
Fixed with the latest TFT release now available on the github.

closed 2020-11-26 by looxonline