looxonline/Marlin #2: [BUG] PID Autotune not working

Ejay-3D, 2020-11-19:
### Bug Description

when running PID autotune from the touchscreen icon 
It start and throw 2 messages : 
1- Unknown command : "M42 P5 S255"
very quickly show ( see the 1st video ) 
then the second message : 
2- Unknown command :"M42 P6 S0"

Then the PID process start . 
When finishing we have a first screen that is shown very quikly and overriden by the next one . 
Process finished Save parameter to EEPROM : YES / NO . But it is not clickable as the second message comes right away : 
Unknown command " M42 P6 S0"
See 2nd video 

### My Configurations

Latest firmware from your repo  BIQUB1_STOCK & ABL 

### Steps to Reproduce

Run AutoPID tunning from touch screen , increase temp to 200 and start. 

**Expected behavior:** [What you expect to happen]

No error message M42 at begin and end , which should allow to save the PID values . 

**Actual behavior:** [What actually happens]

Error messages at the beginning , and at the end which prevent saving the PID tuning result . 

#### Additional Information
[pid.zip](https://github.com/looxonline/Marlin/files/5566497/pid.zip)

looxonline, 2020-11-20:
Will be fixed in the next release.

looxonline, 2020-11-26:
Fixed in the latest TFT release now available on the github.

closed 2020-11-26 by looxonline