looxonline/Marlin #3: ABL mesh (on super res) does not test far end of the bed

DaStivi, 2020-12-19:
Hi, 

ABL leveling starts the test really far in the front of the bed... Befor the leveling screws.

But in the back it stops way before the back end, I would say either one more row would be possible or the testing grid should be slightly shifted backwards

DaStivi, 2020-12-19:
[image]
This is the farest back "line" it's testing...

thisiskeithb, 2021-03-17:
Further back & the nozzle will hit the bed connector/cable when probing that rear-left corner.

DaStivi, 2021-03-17:
> Further back & the nozzle will hit the bed connector/cable when probing that rear-left corner.

no really... only on the newer machines with the cable support strap.. and its even hitting this when using the UBL Firmware from luke allready...

pixeldoc2000, 2021-04-11:
There is also the hardware limitation of the movement of the x-axes, as the Bed can't move much further back for the probe to reach the end.

With the BLTouch Mount from thingiverse right beforce the center Hotend Fan, the Probe is monted a little bit further back.