looxonline/Marlin #5: [BUG] Resume print pop up Label issue in french language.

Ejay-3D, 2021-02-12:
### Bug Description
When a print is resumed after a power outage, when restaring the print the label for the pop up window : 
label_power_failed:Continuer l'impression ?

and 

label_continue:Continuer

Are incorect , those 2 label use the label : 
label_cancel:Annuler
instead . 

[image]

Issue found in All branches 

### Steps to Reproduce
Use this french ini file and flash the screen : 

[language_fr.ini.zip](https://github.com/looxonline/Marlin/files/5970726/language_fr.ini.zip)

**Expected behavior:** [What you expect to happen]
Have the right label for pop up windows and confirm button .

**Actual behavior:** [What actually happens]

See picture higher

DaStivi, 2021-03-17:
this (was) an bug in TFT Firmware, @looxonline  has been fixed ah while ago.. has also been reported on TFT Github ... but cant find the fix or issue right now..

looxonline, 2021-03-17:
> this (was) an bug in TFT Firmware, @looxonline has been fixed ah while ago.. has also been reported on TFT Github ... but cant find the fix or issue right now..

A stable version will be released this weekend. Once released I'll add it to my git. I don't like working with the unstable versions because they're.....unstable 😉

DaStivi, 2021-03-17:
@looxonline i know, i know... i still use the nightly builds.. and besides some issues they work great.... there really where quite some changes since the last "stable"

JaVelin-X, 2021-03-17:
So after I've solved some issues I had with my BLtouch sensor I wanted to add this firmware to my B1. I didn't want to do it sooner because I was worried if the sensor stopped responding it would crash into the bed. So I wonder if I should wait for this update. I'm hoping it solves some other issues I have

Ejay-3D, 2021-03-17:
Thanks Luke, 

Looking forward for the issue fic that I was not able to find out .