looxonline/Marlin #7: TFTFirmware-2020-12-06.zip rename BIGTREE_TFT35_V3.0_E3.26.x.bin to BIGTREE_TFT35_V3.0_B1.26.x.bin

pixeldoc2000, 2021-03-21:
Thanks @looxonline for your awesome work.

I just noticed in TFTFirmware-2020-12-06.zip (all branches) the Firmware File for the Touchscreen is called BIGTREE_TFT35_V3.0_E3.26.x.bin not BIGTREE_TFT35_V3.0_B1.26.x.bin like in the BTT Repo https://github.com/bigtreetech/BIGTREETECH-TouchScreenFirmware .

Maybe "fix" the filename further releases?

looxonline, 2021-04-05:
Good point. I compiled it for the E3 screen which is identical in hardware and the name was generated automatically. I'll tweak the name for the next release which should be soon as the beta was just pushed.

JaVelin-X, 2021-05-04:
I added a smart Filament sensor to my B1 running the insane res firmware and it always signals a runout within 40 seconds of starting to print.  If I turn the smart off in the menu the print resumes perfectly.  I see that the sensor is really connected to the TFT and not the motherboard and I'm not sure how to diagnose it. is this enabled in the TFT firmware somehow? is there a way to check?

pixeldoc2000, 2021-05-05:
> I added a smart Filament sensor to my B1 running the insane res firmware and it always signals a runout within 40 seconds of starting to print. If I turn the smart off in the menu the print resumes [...]

Please open a new Issus. This is an (almost) completly different Topic.

JaVelin-X, 2021-05-05:
> @JaVelin-X
> 
> > I added a smart Filament
> Please open a new Issus. This is an (almost) entirely different Topic.

 Sorry yes, I have. This was the only thread I saw where the TFT was discussed.