looxonline/Marlin #8: Baby steps dont save

Decoydk, 2021-03-24:
Hi

When I adjust my baby steps, and push save icon... It dont save... I used your firmware for plain stock no abl or nothing

michal-szkutnik, 2021-03-25:
Same issue.

pixeldoc2000, 2021-03-25:
Same issue.

Decoydk, 2021-03-31:
> Same issue.
> 
> Maybe this is because `#define EEPROM_AUTO_INIT` is enabled in `Configuration.h` ? Looks like there are some people having simular issus if enabled.
> 
> Noticed it is also enabled in [BTT B1 Firmware](https://github.com/bigtreetech/Marlin/blob/B1-2.0.6/Marlin/Configuration.h) source. I don't remember if it was working with the stock BTT Firmware.
> 
> EDIT: recompiled firmware without `#define EEPROM_AUTO_INIT`, does not work either.
> 
> I don't quite know how this works in marlin and if it is supposed to save "as is". There are a lot of possible settings around Babystepping, Z-Offset, Homeing offset etc.
> 
> Sidenote: the BTT SKR v1.4 hat no dedicated EEPROM. Marlin use EEPROM Emulation...

Maybe you are right ? .... Dont on the stock firmware I think it saved the value

looxonline, 2021-04-05:
Without a probe it is not supposed to allow babystepping. Think through the process...your x gantry lowers until it hits the Z endstop. The endstop is the lowest point that the x gantry can go. If you try to babystep it lower you are simply ramming it into the endstop switch which is putting all kinds of nasty mechanical forces on your machine.

Decoydk, 2021-04-06:
> Without a probe it is not supposed to allow babystepping. Think through the process...your x gantry lowers until it hits the Z endstop. The endstop is the lowest point that the x gantry can go. If [...]

Okay babysteps works on it also on the stock and on other printers I have without probe and they save the babysteps.
When I use babysteps in do NOT hit the endstop.

looxonline, 2021-04-06:
> Okay babysteps works on it also on the stock and on other printers I have without probe and they save the babysteps.
> When I use babysteps in do NOT hit the endstop.

Think about it. If you have homed the z axis and you babystep downwards you ARE hitting the z endstop. There is no way that you can't be. When you calibrate your bed leveling you calibrate it so that at z=0 there is a paper thickness between the nozzle and the bed. At that point the gantry is already hitting the endstop. Take it lower and you're forcing it into the endstop even more. There is no way around it. In a future update I'll look at enabling it only for positive babystepping. The stock firmware allows it because it has been hacked together to have one firmware that supports BL touch and no BL touch so it has no clue if there is a BL touch connected or not when you try to babystep. If it worked correctly then it would stop you from doing it. I don't care how many printers you have where you have tested it, if they allow negative babystepping without a probe then they are doing it wrong and should change their firmware. Only positive babystepping should be allowed.

Decoydk, 2021-04-07:
> > > Without a probe it is not supposed to allow babystepping. Think through the process...your x gantry lowers until it hits the Z endstop. The endstop is the lowest point that the x gantry can go. If you try to babystep it lower you are simply ramming it into the endstop switch which is putting all kinds of nasty mechanical forces on your machine.
> > 
> > 
> > Okay babysteps works on it also on the stock and on other printers I have without probe and they save the babysteps.
> > When I use babysteps in do NOT hit the endstop.
> 
> Think about it. If you have homed the z axis and you babystep downwards you ARE hitting the z endstop. There is no way that you can't be. When you calibrate your bed leveling you calibrate it so that at z=0 there is a paper thickness between the nozzle and the bed. At that point the gantry is already hitting the endstop. Take it lower and you're forcing it into the endstop even more. There is no way around it. In a future update I'll look at enabling it only for positive babystepping. The stock firmware allows it because it has been hacked together to have one firmware that supports BL touch and no BL touch so it has no clue if there is a BL touch connected or not when you try to babystep. If it worked correctly then it would stop you from doing it. It does not matter how many printers you have where you have tested it, if they allow negative babystepping without a probe then they are doing it wrong and should change their firmware. Only positive babystepping should be allowed.

I can see the problem now. But positive babysteps would be great to have better than none... Sure I can but a bl touch on but my printer is straight so I dont see that it is needed for this printer.

pixeldoc2000, 2021-04-11:
> Think about it. If you have homed the z axis and you babystep downwards you ARE hitting the z endstop. There is no way that you can't be. When you calibrate your bed leveling you calibrate it so [...]

@looxonline You have a valid point there. Didn't though about it this way ;-)