looxonline/Marlin #14: [BUG] Double X Home needed

nicedevil007, 2021-07-17:
After installing your firmware and adjusting the sensorless homing sensitivity I figured out that X needs always a 2nd home to home properly. On the first home it only goes a few mm to the oposite direction and then stops there. if I do an other G28 X it will home properly.

nicedevil007, 2021-07-26:
@looxonline So did your calibration guides again. Checked the belts for tension, increased the number for X on M914 until it I could successfull home X multiple times (10 or even 15 times tested).

Then I started a print, first command is G28... it would start to home (X is moving 3 mm away from its actual position) and then moved 3mm in the direction to 0 and stops there.... I do the test again with M914 everything is fine....

Can this be an issue on the stepper current?

I have a heavy modified Sidewinder X1 with an SKR 1.4 turbo, TMC2209 and Klipper running. The 3mm bump back isn't configured there and it is homing all day long. Would it be possible to get a testversion without that for marlin?

looxonline, 2021-07-26:
The back bump is needed because if one tried to home when the axis was already at the endstop it would just grind endlessly. If you are having issues with the X axis then it is very likely that your pulley on the X motor is eccentric. I suggest removing it and studying it very closely to see. The eccentricity causes variable tension along the x travel which messes with the homing. There were some BX units that had this issue, mine was in fact one of them but I just popped a new pulley on there.